vial-kb/vial-gui #18: Delay in Macros possible

JackPikatea, 2021-03-06:
Would it be possible to add a delay or sleep function when creating macros?

xyzz, 2021-03-06:
it's currently not possible due to VIA macro implementation. See here: <https://github.com/vial-kb/vial-qmk/blob/c9c90c06b798f22591b98f8e0de34bde297c6ef5/quantum/dynamic_keymap.c#L336>

Different macros (up to 16) are split with a NUL terminator. Then, in <https://github.com/vial-kb/vial-qmk/blob/c9c90c06b798f22591b98f8e0de34bde297c6ef5/quantum/dynamic_keymap.c#L381> the individual macro is processed. At every character it checks for a magic value:

```
#define SS_TAP_CODE 1
#define SS_DOWN_CODE 2
#define SS_UP_CODE 3
```

and if there isn't one, just treats it as a keycode. Note that the actual keycodes values are as follows:

```
    KC_NO = 0x00,
    KC_ROLL_OVER, // 1
    KC_POST_FAIL, // 2
    KC_UNDEFINED, // 3
    KC_A, // 4
```

which means if we tried to use `#define SS_DELAY_CODE 4` it would conflict with `KC_A`.

So the macro feature would need to be rewritten in order to support delays, probably utilizing a uniform prefix (as QMK does for its own macros). e.g. instead of having 1,4 to mean "tap KC_A", it would be 1,1,4. Then we could use 1,4,XXX for delay (conversely, as mentioned above 4,XXX is ambiguous)

xyzz, 2021-03-06:
Also for implementation of this feature it would help to know what kind of use case you have. So if you have any examples where macro delays are useful, please post them in this thread.

Croktopus, 2021-03-06:
chiming in here cuz delays is a good feature imo - at my work there are pre-defined series of key combos that get interrupted by the server having to process stuff or print, so being able to add like a 2s delay after a keystroke before resuming the macro would be useful.

HellsPlumber, 2021-03-09:
I was hoping to use macros to speed up my work flow when world-building on online platforms such as SecondLife.
I have to repeatedly fill out the same information over and over in forms and on settings pages. 
Every action needs to have a delay between them while the server updates the contents of the form.
As this isn't implemented, the data is entered quicker than the forms/settings can handle them, making my use case unviable.

xyzz, 2021-03-11:
There's a WIP implementation available in https://github.com/vial-kb/vial-gui/tree/macro-delay & https://github.com/vial-kb/vial-qmk/tree/macro-delay - try it out and let me know if that works for your use case.

JackPikatea, 2021-03-11:
I don't see anything different with the GUI. I can't seem to find where to incorporate the delay.

xyzz, 2021-03-11:
You have to compile the firmware from the macro-delay branch, then in gui there is an additional option next to text/down/up/tap.

JackPikatea, 2021-03-16:
Seems to work really well! I did find an issue with setting the delay to blank. This error was shown 
[image]

xyzz, 2021-03-24:
I've made a bunch of updates to this feature, could you please test latest builds of vial-gui as well as vial-qmk. The same branches as before:
- https://github.com/vial-kb/vial-gui/tree/macro-delay
- https://github.com/vial-kb/vial-qmk/tree/macro-delay

xyzz, 2021-04-03:
Totally forgot to close this, it is now fixed with the v0.2 release.

closed 2021-04-03 by xyzz