vial-kb/vial-gui #19: Rotary encoders cannot use macros

monkeychou, 2021-03-20:
The encoder uses a macro, but there is no response
I'm not good at English

xyzz, 2021-03-20:
Please upload your QMK source files including all configuration & Vial JSON.

monkeychou, 2021-03-20:
this is my source files

[newpad.zip](https://github.com/vial-kb/vial-gui/files/6175934/newpad.zip)

monkeychou, 2021-03-20:
[vial.json.zip](https://github.com/vial-kb/vial-gui/files/6175938/vial.json.zip)

xyzz, 2021-03-20:
It looks like the same issue we've seen before, it is running out of stack memory and clashing with the global data, corrupting random data (you can confirm it by disabling `RGBLIGHT_ENABLE` and then macros execute ok). Though I am not sure why it would run out in this particular case as it looks like there should be plenty of memory available.

xyzz, 2021-03-21:
After looking into it more, it is actually caused by a memory corruption when layer cache is being updated. So I've fixed it in vial-qmk `encoder-simplify` branch, please try it and let me know if it resolves the issue: https://github.com/vial-kb/vial-qmk/tree/encoder-simplify

monkeychou, 2021-03-22:
Yes, the encoder can use macros, but will touch off twice。

https://user-images.githubusercontent.com/63698796/111965618-e44ee980-8b30-11eb-9b0d-8c7ea358adc6.mp4

xyzz, 2021-03-22:
ah my bad, try updating the branch I just fixed that

monkeychou, 2021-03-22:
> ah my bad, try updating the branch I just fixed that

Perfect. Thank you very much

xyzz, 2021-03-22:
fixed in https://github.com/vial-kb/vial-qmk/commit/c69a05c2167c2ce6d7b798b1fea05f5d7b3bb75a

closed 2021-03-22 by xyzz

anubhavd7, 2022-08-12:
@xyzz hello sir i want to make a macropad with just one rotary encoder. i am facing a problem in it the button key is working perfectly but the rotary part clockwise and anticlockwise is not working. I have tried alot of things but nothing is working out can you please help me in it