vial-kb/vial-gui #23: Feature request: Test Matrix feature

Pieterv24, 2021-04-04:
One of the things I currently miss in the GUI, is the ability to test whether keys work and are pressed correctly acoarding to the matrix

xyzz, 2021-04-04:
Actually I've removed that functionality from vial-qmk because it's essentially a keylogger. So it's not possible to implement a matrix test, it is however possible to implement a basic keycode test.

Pieterv24, 2021-04-07:
would it be possible to reintruduce it, behind the security keys feature?
It was something I thought was quite handy.
I'm unsure how it was implemented in QMK in the first place though

Nisker, 2021-04-07:
> I'm unsure how it was implemented in QMK in the first place though
> 
I think it can be reimplemented in your firmware (vial-qmk) by commenting out or deleteing the "goto skip;" on line 260 in "quantum/via.c". 
Then recompile and flash your kb and use VIA to test matrix and vial for everything else.
_disclaimer: I haven't tested it, and might be wrong_

Do note it brings back the security risk, that a malicious program could monitor your keys in the background.

xyzz, 2021-04-23:
Merged the PR, thanks for contributing

closed 2021-04-23 by xyzz