vial-kb/vial-gui #31: "Download VIA definitions" gives an SSL error

marksdk, 2021-05-19:
**Environment**

macOS 11.2.3, M1 MacBook Pro

**Steps to reproduce**

1. Open Vial
2. Click "Download VIA definitions"
3. See error message

**Error message**

```
  File "main_window.py", line 251, in load_via_stack_json
  File "urllib/request.py", line 223, in urlopen
  File "urllib/request.py", line 526, in open
  File "urllib/request.py", line 544, in _open
  File "urllib/request.py", line 504, in _call_chain
  File "urllib/request.py", line 1392, in https_open
  File "urllib/request.py", line 1351, in do_open

URLError: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:852)>
```

xyzz, 2021-05-22:
Could you check this build and let me know if it fixes the issue - https://github.com/vial-kb/vial-gui/actions/runs/867529496

marksdk, 2021-05-24:
@xyzz Yes, that's perfect! No errors during loading of VIA definitions, and my Thermal layout loaded perfectly too :👌

closed 2021-05-24 by marksdk