vial-kb/vial-gui #39: Move away from fbs

Kranzes, 2021-07-05:
Hey I was wondering if there was a possibility that this project will move away from using fbs in the near future since fbs is still using python 3.6?

Kranzes, 2021-07-05:
Also updating PyInstaller to a version which isn't 2+ years old, fbs also depends on this outdated version of PyInstaller sadly.

xyzz, 2021-07-05:
The plan is to eventually switch to using pyinstaller directly, unsure when I would get time to work on that though.

Kranzes, 2021-07-05:
Alright so i had a discussion with the author of fbs and he doesn't have any plans on updating to anything over python 3.6, if you want to get the a newer python version which isn't insecure according to Github, you will have to pay $80 for his closed-source software that is smaller than the damn website for it. The developer seems to be very against FOSS/Open-Source.

xyzz, 2021-07-05:
yes it was very disappointing that the original open source version was licensed under GPL, yet apparently there is no way to use the updated one within a GPL project

xyzz, 2021-07-06:
The plan is to switch away from fbs to pyinstaller in 0.5 (so, the release after the next one).

branch to track changes: https://github.com/vial-kb/vial-gui/tree/pyinstaller

Current status:
- Linux: appimage fails with "qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found."
- Windows: nothing yet
- macOS: nothing yet

dwlocks, 2024-04-17:
I created https://github.com/vial-kb/vial-gui/pull/238 for two reasons:
1. The available latest download showed text too tiny to read in the main window (both the refresh button and the no keyboard found message). Menu text was normal.
2.  My keyboard was not recognized, and I hoped updating some libs would help.

This branch updates to python 3.12, updates to latest qt5, and dependencies, removes fbs and makes things run.

It fixes my tiny text problem (1). It did not help with recognizing my keyboard (2).

I'll make whatever changes you want so you can merge these changes. I didn't test on linux or mac.  I'm going to optimistically predict that linux will build with the new dependencies. :-)