vial-kb/vial-gui #44: The Tap Dance tab is empty

georgezhe, 2021-09-07:
I'm on macOS Catalina 10.15.7 and when I open vial for my Melody65 keyboard, it detects the keyboard successfully, but the tap dance tab is empty.

Tubo, 2021-09-27:
Same on KDE Neon with AppImage.

xyzz, 2021-09-28:
While there's a bug here (the empty tab should be hidden altogether), it's probably not the fix you want. If you're looking for tap dance support in vial gui, you must be running latest vial-qmk firmware (not via firmware).

Tubo, 2021-09-28:
That solved it. For future reference, I followed the 'porting guide' and installed the vial-qmk firmware. Now there are many more options showing up, including the tap/hold-related options. Awesome!

xyzz, 2021-09-30:
Fixed by 8db625e3804dd2460b4ecc06ad6d01a6ff153eff

closed 2021-09-30 by xyzz