vial-kb/vial-gui #48: Layout options with keys/encoder above the top row (OSX)

BenRoe, 2021-09-26:
Hello,
i have a Macro Pad (4x4) with optional encoder (top row, or right row).
I try to add an option to switch the encoder position. From "top" to "right".

This works on Windows, but not on Mac OSX (10.15) (VIAL v0.4.1)

Not related, but is there a reason why the layout section is in new tab and not like in the window version?

```json
{
    "name": "Plaid-Pad",
    "vendorId": "0x7983",
    "productId": "0x5050",
    "lighting": "none",
    "matrix": {
        "rows": 4,
        "cols": 4
    },
    "layouts": {
      "labels": [
        "Encoder TOP",
        "Encoder RIGHT"
      ],
      "keymap": [
        [{"x":2},"0,1\n\n\n0,1\n\n\n\n\n\ne","1,1\n\n\n0,1\n\n\n\n\n\ne","2,1\n\n\n0,1\n\n\n\n\n\ne","3,1\n\n\n0,1\n\n\n\n\n\ne"],
        [{"x":2},"0,0\n\n\n0,1\n\n\n\n\n\ne","1,0\n\n\n0,1\n\n\n\n\n\ne","2,0\n\n\n0,1\n\n\n\n\n\ne","3,0\n\n\n0,1\n\n\n\n\n\ne"],
        [{"x":2},"1,0","1,1","1,2","1,3",{"x":0.5},"3,0\n\n\n1,1\n\n\n\n\n\ne","3,1\n\n\n1,1\n\n\n\n\n\ne"],
        [{"x":2},"1,0","1,1","1,2","1,3",{"x":0.5},"2,0\n\n\n1,1\n\n\n\n\n\ne","2,1\n\n\n1,1\n\n\n\n\n\ne"],
        [{"x":2},"2,0","2,1","2,2","2,3",{"x":0.5},"1,0\n\n\n1,1\n\n\n\n\n\ne","1,1\n\n\n1,1\n\n\n\n\n\ne"],
        [{"x":2},"3,0","3,1","3,2","3,3",{"x":0.5},"0,0\n\n\n1,1\n\n\n\n\n\ne","0,1\n\n\n1,1\n\n\n\n\n\ne"]
      ]
    }
}
```

https://user-images.githubusercontent.com/643834/134818145-2aa94c12-74a2-4815-9135-fc94dcf50c54.mov

xyzz, 2021-09-28:
Does that still happen with the latest build from https://github.com/vial-kb/vial-gui/actions/runs/1215427761?

BenRoe, 2021-09-28:
With the gui version from the linked action it crash with the json from above

https://user-images.githubusercontent.com/643834/135107551-00fcfe7c-69c7-419e-b85c-7720e7da9d51.mov

.

xyzz, 2021-09-30:
Ok, I've looked over your layout and the issue is that it's malformed - you have entries for e.g. "0,1" and "1,1" but not "0,0" and "1,0" layout options. This is not supported. You have to add blank/decal keys, then it works correctly, here's an example:

[issue-48.txt](https://github.com/vial-kb/vial-gui/files/7261715/issue-48.txt)

>  Is it possible to switch clockwise and counter clockwise in the gui for the encoder?

No, you have to change that in QMK config (swap your A and B pins for the encoder).

closed 2021-09-30 by xyzz