vial-kb/vial-gui #56: Tap dance configuration tabs are missing

trackballenjoyer, 2021-11-11:
I'm new to VIA and Vial, so this could totally have something to do with my keyboard the sideloaded layout I'm using. I'm not sure, but the tap dance configuration tabs are missing where the documentation indicates they would be:

### From the docs
[image]

### What I see
[image]

### Details
Operating system: macOS Monterey 12.0.1
Vial version: 0.4.1

NCKiser, 2021-11-11:
Do you have Tap Dance enabled on the keyboard's firmware? If the keyboard is running VIA firmware, Tap Dance is not currently configurable with the Vial GUI.

trackballenjoyer, 2021-11-11:
Tap Dance and VIA are both enabled for my firmware. I assumed I would need both for Vial to work and for Tap Dance to be configurable with it.

The documentation seems to suggest that Tap Dance *is* configurable with Vial: https://get.vial.today/manual/tap-dance.html#tap-dance. Is this not fully implemented yet?

NCKiser, 2021-11-11:
It is configurable with Vial, when the keyboard is running Vial firmware. Your keyboard is running VIA firmware.

You can port from VIA to Vial by following the porting guide here:
https://get.vial.today/docs/porting-to-vial.html

trackballenjoyer, 2021-11-11:
Gotcha. I assumed Vial functioned more as an alternative frontend for VIA with some additional configuration options. I didn't realize that it required different firmware, but that makes sense.

Thanks!

closed 2021-11-11 by trackballenjoyer