vial-kb/vial-gui #57: Vial windows Error

a151542a, 2021-12-01:
I have modified and successfully compiled Vail's qmk firmware according to the instructions, and flashed into my diy keyboard.
When I plug in the keyboard and open the Vial software in win11, Vial software reports an error：
File”main——windos.py“, line  252, in on_device_selected
File"vial_device.py", line 50, in open
File"keyboard_comm.py", line 225, in reload
File"keyboard_comm.py", line 311, in reload_layout
ValueError:invalid literal for int() with base 10:"7"

But，If I don’t plug in the keyboard, it will not prompt

xyzz, 2021-12-01:
it indicates an issue with your vial.json, I suggest that you ask on our discord for support - https://discord.gg/6Ybrtvj6Ae

closed 2021-12-01 by xyzz

a151542a, 2021-12-01:
> 

But this via.json works fine in VIA

a151542a, 2021-12-01:
> 这证明您的 vial.json 问题，我建议您在我们存在的不善处寻求支持 - https://discord.gg/6Ybrtvj6Ae
But this via.json works fine in VIA