vial-kb/vial-gui #59: KC_MUTE vs KC__MUTE

grizleeeee, 2021-12-15:
In a follow up to this issue: https://github.com/vial-kb/vial-gui/issues/45

`KC_MUTE` only enables Mute on macOS (12.1), it doesn't toggle it.  However, `KC__MUTE` does toggle it.  

When `KC__MUTE` is set from QMK, this works without an issue, but when I open VIAL, it overwrites it.  When saving that layout and then opening it in a text editor, the value was replaced with `127`.  When I then load said saved layout, it acts the same as `KC_MUTE`, thus only enabling Mute, not disabling it if it's already enabled.

Could `KC__MUTE` be added to the GUI?  Perhaps denoting that it will only work for macOS and Linux?

xyzz, 2021-12-15:
Yes `KC__MUTE` can be added, however I cannot reproduce the issue with Vial overwriting a keycode, can you upload a video perhaps that showcases the issue?

grizleeeee, 2021-12-15:
Please ignore my incompetence.  This isn't an issue.  If the Volume was all the way at 0 when `KC_MUTE` was pressed, it just reapplies Mute, however, when volume is above 0, it will toggle between the two, this works with both of them.  This is a non-issue.  100% user error.

closed 2021-12-15 by grizleeeee