vial-kb/vial-gui #67: Documentation experience feedback

benfrain, 2022-01-14:
Firstly thanks for making this tool - looks like just the thing I want if I could get it working. 

Just commenting here in the hope of providing some useful feedback for the docs. Obviously this is just one user but maybe my 'fresh visitor' experience will be useful. 

# Background
Built Sofle KB and flashed with QMK, then read about Vial on Reddit as a GUI for easier tweaking of layout etc. Came to the Vial website with a view to installing and using.

# Experience
Read the First Time User section...

- I download Vial (step 1)
- I run Vial (Step 2)

Vial starts and it shows this:

[image]

And obviously Step 3 is then not possible.

I try the 'Download VIA definitions' from the menu and nothing happens/changes. I restart. Same situation. Do I need to put my board into DFU mode or something?

Unsure what to do next I head to the 'Porting a keyboard to Via' section linked from the top of the First Time Use page. 

Then further confusion. Top of the page states: 'vial-qmk provides several examples of most common microcontroller configurations set up for Vial usage. You can check them out at vial-qmk/keyboards/vial_example.' say what? Where is that location and do I need to be getting a micro controller configuration?

Then reading down the page we are onto cloning the Vial QMK fork and creating a via keymap.... and at this point I'm done.

# What I felt I needed
As a newcomer, and maybe I'm not the intended audience, but what I want to know is...

- What is VIA? Why is it continualy mentioned? Why do I care about that? What has Via to do with Vial?
- Are any keyboards supported out of the box with Vial? If they are, what are they? Is there a list? How do I know if my board is 'Vial-compatible'?
- If my keyboard isn't supported out of the box, what steps are necassary to add support (I think this is the existing 'Porting' section)?
- What are the pre-requisites for using Vial? Do I need a JSON of some sort? Do I need a fork of QMK already on my machine to run with this?

xyzz, 2022-01-14:
To be honest I think the issue is with whoever told you to use Vial without providing any more details, as I believe we're pretty clear about compatibility (Vial-compatible device, VIA keyboards) - maybe that could be clarified further?

Now to answer the questions:
> What is VIA? Why is it continualy mentioned? Why do I care about that? What has Via to do with Vial?

VIA is the original closed-source real-time keyboard configuration GUI for QMK, both VIA and Vial share a very similar protocol for configuring keyboard. However as an end user what you need to know is that Vial is compatible with both Vial and VIA keyboards.

> Are any keyboards supported out of the box with Vial? If they are, what are they? Is there a list? How do I know if my board is 'Vial-compatible'?

Yes, a few ship with Vial firmware and much more with VIA, there isn't a centralized list since well a part of Vial design is that there's no centralization and that anybody can port any boards they'd like without having to obtain our approval (unlike in VIA). Typically you'd find that information on the vendor website where you're purchasing the keyboard, some examples I know of are Matrix ME, Pikatea macropads, kbdfans also used to ship some of their boards with Vial, and there's been some 40% keyboards.

> If my keyboard isn't supported out of the box, what steps are necassary to add support (I think this is the existing 'Porting' section)?

The steps are detailed in the porting section - https://get.vial.today/docs/ - however this is not intended to be followed by the end-user, but rather by the keyboard designer (of course you can still follow these steps and port a board, however we assume the readers are familiar with common QMK firmware terms)

> What are the pre-requisites for using Vial? Do I need a JSON of some sort? Do I need a fork of QMK already on my machine to run with this?

The pre-req for using Vial is a keyboard flashed with a Vial firmware, you could also use a VIA firmware with reduced functionality. The pre-reqs for porting are detailed in the porting guide, but generally the keyboard must run QMK, and then you need a VIA json.

tl;dr - the intent is that the keyboard vendor pre-flashes the boards with Vial firmware (or provides a Vial firmware download), so perhaps there's a mismatch in expectations.

benfrain, 2022-01-14:
Appreciate the information. 👍

closed 2022-01-14 by benfrain