vial-kb/vial-gui #70: No AltGr option as keymap

Adxm03, 2022-01-19:
Cannot find the AltGr key anywhere.

this keymap in qmk is KC_ALGR.

AltGr is essential for my studies so I was upset to see it isnt supported properly afaik

Nisker, 2022-01-19:
AltGr is not a part of the normal US layout, but its equivalent to  the right alt (RAlt). The KC_ALGR is just an alias https://github.com/qmk/qmk_firmware/blob/master/docs/keycodes.md

If you use one of the regional layouts from the menu in the top of vial and its incorrect, the creator might have missed it. But that said its just visual, functionally Ralt and AltGr are the same.

closed 2022-01-23 by xyzz

MarkuBu, 2024-11-03:
That's not true. RAlt and AltGr are different. But I found a strange behavior in VIAL (and RMK as firmware). On my keyboard the LAlt key acts as AltGr. I don't know why. The RAlt works as LAlt.