vial-kb/vial-gui #84: Mouse key settings missing from QMK settings in v0,4.2. Feature regression?

ujl123, 2022-03-24:
https://get.vial.today/changelog/release-0.4.html#mouse-keys

The docs lists the mouse-keys settings but with the newest version, I do not  see this option anywhere. My version also doesn't  include the "Firmware updater". Both the windows  installer version and the windows portable v.0,4,2 don't have it.

[image]

xyzz, 2022-03-24:
Mouse keys tab missing indicates the firmware was compiled without qmk mousekey support.

Firmware updater tab is missing because that's only included for keyboards utilizing the [vibl](https://github.com/vial-kb/vibl) protocol, which gmmk pro isn't using.

closed 2022-03-24 by xyzz

ujl123, 2022-03-24:
> Mouse keys tab missing indicates the firmware was compiled without qmk mousekey support.
> 
> Firmware updater tab is missing because that's only included for keyboards utilizing the [vibl](https://github.com/vial-kb/vibl) protocol, which gmmk pro isn't using.

Thank you for the confirmation!