vial-kb/vial-gui #104: GUI not showing up

Stunkymonkey, 2022-05-28:
I am using Sway (Wayland) on NixOS.
When starting `Vial` the GUI does not show up.
The following errors are shown:
```
QXcbConnection: XCB error: 2 (BadValue), sequence: 479, resource id: 1024, major code: 130 (Unknown), minor code: 3
QXcbConnection: XCB error: 2 (BadValue), sequence: 482, resource id: 1024, major code: 130 (Unknown), minor code: 3
QXcbConnection: XCB error: 2 (BadValue), sequence: 486, resource id: 1024, major code: 130 (Unknown), minor code: 3
```
Any idea, what is causing this?
I did install the `0.4.1` and the `0.6`. Same results.

Ding-Fan, 2023-01-17:
I met similar issue on Windows11. It was showing up at first, several days later, suddenly the GUI won't showing up.

Stunkymonkey, 2023-12-28:
no longer an issue for me. Somehow got fixed.

closed 2023-12-28 by Stunkymonkey