vial-kb/vial-gui #105: Space key not shown in VIAL UI for Keychron sideloaded JSONs

CrazyCoder, 2022-05-28:
Sample JSONs:

https://cdn.shopify.com/s/files/1/0059/0630/1017/files/q1_us_ansi_non_knob_ver1.08.json.zip?v=1650248055
https://cdn.shopify.com/s/files/1/0059/0630/1017/files/q2_us_ansi_non_knob_ver1.04.json.zip?v=1650248870

These JSON files are linked from the following pages:

https://www.keychron.com/blogs/archived/how-to-reset-your-keychron-q1-to-factory-settings
https://www.keychron.com/blogs/archived/how-to-factory-reset-or-flash-your-qmk-via-enabled-keychron-q2-keyboard

VIAL does not display the Space button in the UI:

[image]

VIA works fine:

[image]

xyzz, 2022-05-28:
It's a bug in keychron's json:
[image]
The key location should be a top left legend, not the center one. It's mentioned in VIA docs as well <https://www.caniusevia.com/docs/layouts#switch-matrix-co-ordinates>

It's currently not a goal to be bug-compatible with VIA so there's no plans fixing it.

closed 2022-05-28 by xyzz

CrazyCoder, 2022-05-29:
Thanks for the pointer, I've managed to fix the JSON to display the Space key.