vial-kb/vial-gui #106: v0.6 Hangs on Launch - macOS 12.3.1 on Apple Silicon

IanOfGit, 2022-05-30:
Downloaded Vial DMG from the site, moved to Applications Folder, double clicked to launch after giving security permission to launch.

Icon just bounces on the dock and goes unresponsive. Persists after reboot.

Plandercozeron, 2022-07-13:
Somewhat same behaviour. But keyboard can't be detected
Macbook Pro 14'' on M1 Pro.
MacOS 12.4

1) Flashed Planck rev6 with via binary
2) Launch Vial. It hang for a little
3) **File -> Download VIA definitions** OR hit **Refresh**
4) I get an error:

```
File "main_window.py", line 270, in on_devices_updated
File "main_window.py", line 274, in on_device_selected
File "autorefresh/autorefresh.py", line 77, in select_device
File "vial_device.py", line 49, in open
File "vial_device.py", line 26, in open

RuntimeError: unable to open the device
```
for linux there is an instruction for adding device to udev, but MacOS works different.
Any clues?

AndreySypachev, 2022-07-13:
Python 3.9.13

jldohmann, 2022-08-06:
Having the same issue, though I'm on an Intel Mac: vial worked when first downloaded and opened (after granting permission), but on subsequent tries the application hangs

blowmage, 2022-11-08:
Ran into this as well using v0.6 with a wilba.tech wt60-h1 pcb on an older MacBook Pro. Works fine on h60 and nop60 pcbs however. Still have the issue even after downloading via definitions.

```
Vial 0.6
Python 3.6.8
Qt 5.9.3
```

```
OS X 12.6.1
2.5 GHz Quad-Core Intel Core i7
```

```
File "main_window.py", line 270, in on_devices_updated
  File "main_window.py", line 274, in on_device_selected
  File "autorefresh/autorefresh.py", line 77, in select_device
  File "vial_device.py", line 49, in open
  File "vial_device.py", line 26, in open

RuntimeError: unable to open the device
```

vieraleonel, 2024-01-22:
> Downloaded Vial DMG from the site, moved to Applications Folder, double clicked to launch after giving security permission to launch.
> 
> Icon just bounces on the dock and goes unresponsive. Persists after reboot.

Same here. Last time i've used it was on first week of december and it was working fine. Don't know what changed (maybe a patch update).

Is there any way to get some logs about the crash?

I was trying to run it from code but can't get to work the dependencies in m1/arch

GibbyWoyto, 2024-05-09:
I got the same issue. Cant find anything that would help. is ther really no one who knows what this error means?

vieraleonel, 2024-05-20:
If it helps i download source code and do the following changes to make it work (on macos sonoma):

```diff
diff --git a/.python-version b/.python-version
index 40c341b..30291cb 100644
--- a/.python-version
+++ b/.python-version
@@ -1 +1 @@
-3.6.0
+3.10.0
```

```diff
diff --git a/requirements.txt b/requirements.txt
index 682f6c9..bb632ee 100644
--- a/requirements.txt
+++ b/requirements.txt
@@ -1,13 +1,11 @@
-altgraph==0.17
-fbs==0.9.0
-future==0.18.2
 git+https://github.com/vial-kb/cython-hidapi@88d8983b3dc65b9ef84238abe3f46004b0ef2fd0#egg=hidapi
+# fork of fbs that is py 3.10 compatible
+git+https://github.com/rnbwdsh/fbs
+certifi==2022.6.15
 keyboard==0.13.5
-macholib==1.14
-pefile==2019.4.18
-PyInstaller==3.4
-PyQt5==5.9.2
-https://github.com/danthedeckie/simpleeval/archive/41c99b8e224a7a0ae0ac59c773598fe79a4470db.zip
-sip==4.19.8
-pywin32==303; sys_platform == 'win32'
-certifi
+macholib==1.16
+pefile==2022.5.30
+pyqt5==5.15.10
+simpleeval==0.9.12
+pyinstaller==4.10
+pywin32==304; sys_platform == 'win32'
```

and bake a bash file called vial-gui to launch the app:
```bash
#!/usr/bin/env sh
DIR="$(CDPATH= cd -- "$(dirname -- "$0")" && pwd -P)"
$DIR/venv/bin/python $DIR/src/main/python/main.py
```

I also have these changes but i don't remember why ... may it'll make windows work:
```diff
diff --git a/src/installer/windows/Installer.nsi b/src/installer/windows/Installer.nsi
index 2dc6b1f..fb30f55 100644
--- a/src/installer/windows/Installer.nsi
+++ b/src/installer/windows/Installer.nsi
@@ -49,7 +49,7 @@ FunctionEnd
     !define MUI_FINISHPAGE_RUN
     !define MUI_FINISHPAGE_RUN_CHECKED
     !define MUI_FINISHPAGE_RUN_TEXT "Run ${app_name}"
-    !define MUI_FINISHPAGE_RUN_FUNCTION "LaunchLink"
+;    !define MUI_FINISHPAGE_RUN_FUNCTION "LaunchLink"
   !insertmacro MUI_PAGE_FINISH

   !insertmacro MUI_UNPAGE_CONFIRM
@@ -99,7 +99,7 @@ Section "Uninstall"

 SectionEnd

-Function LaunchLink
-  !addplugindir "."
-  ShellExecAsUser::ShellExecAsUser "open" "$SMPROGRAMS\${app_name}.lnk"
-FunctionEnd
\ No newline at end of file
+;Function LaunchLink
+;  !addplugindir "."
+;  ShellExecAsUser::ShellExecAsUser "open" "$SMPROGRAMS\${app_name}.lnk"
+;FunctionEnd
```