vial-kb/vial-gui #108: QMK Settings -> Tap Hold -> Permissive Hold not working

sommerper, 2022-06-17:
Hi and first of all: Thanks for a great app!

I'm trying to configure my Contra so when I hold Enter and tap another key it will act as a Shift modifier.
I was under the impression that Permissive Hold would enable me to ignore the Tapping Term and skip straight to modifying the next key press.

For example.
If I want to write "?" then I hold "ENTER" and wait 200 ms and press "/"
That works.

If I enable Tapping Term and hold "ENTER" less than 200 ms and press "/" it will fire "ENTER" and "/".
Not "?" as expected.

[image]
[image]
[image]

I hope I'm not doing anything stupid.

RichardBronosky, 2022-07-25:
I have the exact same need and experience the exact same bug.

xyzz, 2022-07-26:
Use mod-tap instead of tap dance. I.e. set it to `LSFT_T(KC_ENTER)`

closed 2022-07-26 by xyzz

RichardBronosky, 2022-07-26:
> `LSFT_T(KC_ENTER)`

I've done this, but I still get `d` instead of `D` when (all within less than `TAPPING_TERM`) I do;

1. Press `KC_ENTER`
2. Press `KC_D`
3. Release `KC_ENTER`
4. Release `KC_D`

...which is the natural way we do a finger roll when capitalizing letters. This works fine with a normal `KC_LSHIFT` key. It only works for `KC_ENTER` if after [1] I wait `TAPPING_TERM` before continuing.

RichardBronosky, 2022-07-26:
Please reopen.

xyzz, 2022-07-26:
I cannot reproduce this issue with default tap-hold options.

[image]

nfeddema, 2026-01-03:
Same issue.  I am using a home row mod with alt on 's' and when I press s+tab it should register alt+tab when permissive hold is enabled.  It doesn't.  I am holding 's' while pressing and releasing tab within the tapping term.

The mod-tap suggestion is not helpful as mod-tap can't handle rolls, whereas permissive hold can (i.e. permissive hold is supposed to distinguish between nested key presses and rolled key presses).