vial-kb/vial-gui #116: N-key rollover re-enables itself each time user logs in or reconnects the keyboard on MacOS

Talent-Lassikko, 2022-07-28:
As the title states.

Steps to reproduce:
1. Plug in keyboard
2. Open Vial and navigate to the "QMK Settings" tab
3. Select the "Magic" subtab 
4. Deselect the "Enable N-key rollover" -checkbox and press "Save".
5. Close Vial
6. Unplug and replug the keyboard or log out and log back in
7. Open Vial again and navigate back to the "QMK settings" and "Magic" tab
8. The "N-key rollover" has been re-enabled

After saving the QMK settings:
[image]

After replugging the keybaord and restarting Vial:
[image]

xyzz, 2022-07-28:
Do you have FORCE_NKRO enabled?

vinorodrigues, 2022-10-24:
> Do you have FORCE_NKRO enabled?

Would it not be better to detect if this was done or not?  To me it would make sense if the f/w was not compiled with FORCE_NKRO that it would account for this in the UI.