vial-kb/vial-gui #117: AttributeError: 'NoneType' object has no attribute 'keyboard_id'

jsodeman, 2022-08-01:
I imported a layout using Sideload VIA JSON. Then I customized the layout and saved it using Save current layout

When I try to open the .via file I get an error

[image]

Windows 11
Vial 0.6
Python 3.6.8
Qt 5.9.3

File contents below
===============================================
```
{"version": 1, "uid": -1, "layout": [[["M0", "KC_2", "KC_3", "KC_4", "KC_MPLY", "KC_MNXT", "KC_MPRV", -1], ["KC_5", "KC_6", "KC_7", "KC_8", "TO(1)", "KC_PGUP", "KC_PGDOWN", -1], ["KC_9", "KC_0", "KC_UP", "KC_ENTER", "KC_MUTE", "KC_VOLU", "KC_VOLD", -1], ["M1", "M2", "KC_DOWN", "KC_RIGHT", -1, -1, -1, -1]], [["KC_TRNS", "KC_TRNS", "KC_TRNS", "KC_TRNS", "KC_TRNS", "KC_TRNS", "KC_TRNS", -1], ["KC_TRNS", "KC_TRNS", "KC_TRNS", "KC_TRNS", "TO(2)", "KC_TRNS", "KC_TRNS", -1], ["KC_TRNS", "KC_TRNS", "KC_TRNS", "KC_TRNS", "KC_TRNS", "KC_TRNS", "KC_TRNS", -1], ["KC_TRNS", "KC_TRNS", "KC_TRNS", "KC_TRNS", -1, -1, -1, -1]], [["KC_TRNS", "KC_TRNS", "KC_TRNS", "KC_TRNS", "KC_TRNS", "KC_TRNS", "KC_TRNS", -1], ["KC_TRNS", "KC_TRNS", "KC_TRNS", "KC_TRNS", "TO(3)", "KC_TRNS", "KC_TRNS", -1], ["KC_TRNS", "KC_TRNS", "KC_TRNS", "KC_TRNS", "KC_TRNS", "KC_TRNS", "KC_TRNS", -1], ["KC_TRNS", "KC_TRNS", "KC_TRNS", "KC_TRNS", -1, -1, -1, -1]], [["RGB_SPI", "RGB_SPD", "KC_TRNS", "KC_TRNS", "KC_TRNS", "KC_TRNS", "KC_TRNS", -1], ["RGB_SAI", "RGB_SAD", "KC_TRNS", "KC_TRNS", "TO(0)", "KC_TRNS", "KC_TRNS", -1], ["RGB_TOG", "RGB_MOD", "RGB_HUI", "KC_TRNS", "KC_TRNS", "KC_TRNS", "KC_TRNS", -1], ["KC_TRNS", "RGB_VAI", "RGB_HUD", "RGB_VAD", -1, -1, -1, -1]]], "encoder_layout": [[], [], [], []], "layout_options": -1, "macro": [[["text", "abcdefg"], ["tap", "KC_ENTER"]], [["text", "abcdefg"]], [["text", "abcdefg"]], [], [], [], [], [], [], [], [], [], [], [], [], []], "vial_protocol": -1, "via_protocol": 9, "tap_dance": [], "combo": [], "settings": {}}
```

jsodeman, 2022-08-11:
I tried with another vial file and got the same error. If it's helpful here are a couple log entries.

```
2022-08-11 10:50:58,319 - INFO - util:120 - Matching VID=320F, PID=5044, serial=, path=b'\\\\?\\hid#vid_320f&pid_5044&mi_01&col04#8&15d0abeb&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}' - VIA stack
2022-08-11 10:50:58,319 - WARNING - util:70 - is_rawhid: b'\\\\?\\hid#vid_320f&pid_5044&mi_01&col04#8&15d0abeb&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}' does not match - usage_page=FF00 usage=01
```

jsodeman, 2022-08-11:
I figured out the issue. The kb isn't in the registry so I have to sideload the JSON before I can load the layout.

closed 2022-08-11 by jsodeman

hrw, 2025-10-20:
Still - it should give a message instead of throwing an exception.