vial-kb/vial-gui #122: The layer cannot exceed 17 layers. If the layer exceeds 17 layers, some layer functions will have errors. For example, when qmk has 20 layers, when TO (0) is set, the output is TO (16).

jeromeli002, 2022-08-09:
The layer cannot exceed 17 layers. If the layer exceeds 17 layers, some layer functions will have errors. For example, when qmk has 20 layers, when TO (0) is set, the output is TO (16).

BrianCArnold, 2022-08-16:
I don't know how much this will help, but this sounds very suspiciously like the layer ID passed to TO() is only 4 bits long. This could be because the keycodes are 16 bits long, so for "TO" and "MO", etc, the first 12 bits state what kind of action it is, and the last 4 bits are the layer id. If that's the case, it's going to have an overflow and mess things up. But I haven't looked at the code, so...

xyzz, 2022-08-16:
It's a qmk limitation as mentioned in https://github.com/qmk/qmk_firmware/issues/17960

closed 2022-08-16 by xyzz