vial-kb/vial-gui #123: Keyboards with vertically staggered columns do not follow sane "next key" choices.

BrianCArnold, 2022-08-16:
Example, Keebio Iris keyboard. I'm gonna try to figure out how to fix this myself, and if I can, I'll submit a pull request. Any devs with ideas on where to look, etc, let me know, I'm a C# dev, not Python, but hey, I gotta learn somehow.

BrianCArnold, 2022-08-16:
So, I confirmed the cause of the problem, and I created a pull request as a reference point for the cause of the issue. Just removing that `sort` line would almost certainly cause problems for people using split keyboards or something like that, so I don't think that's the solution. But I think adding a new boolean flag in the firmware, next to the key layout from the JSON for the keymap, that tells the GUI not to sort the keys like it does, would be extremely helpful. I have a keymap for the Keebio Iris I'm going to create a pull request for in a moment, if a dev just has a DFU ATMEGA32U4 laying about, they can flash my Vial keymap for the Iris and connect it to Vial and see what I mean when you assign a key. If holds by Row first, and since the rows aren't straight due to the vertical staggering, it seems to jump randomly.

Another solution would be to have it still sort by Y then X, but only sort by the FLOOR of Y (and possibly X), so that developer could just make sure that the keys are laid out in a way that each "row" from staggered column to column, has the same integer component, then it would behave sanely?

HubKing, 2025-03-15:
Since the app already knows the physical locations of keys, can't it just move to the closest key on the right or on the leftmost key of the next line? Currently, it moves like the screen recording below, which makes me click the 'sane' next key and the key code back and forth.

https://github.com/user-attachments/assets/55e53927-682c-497c-b18a-9bb263e31899