vial-kb/vial-gui #134: Unsupported protocol

GermanG, 2022-09-22:


xyzz, 2022-09-22:
Yes, the latest VIA protocol is currently not supported. You should use the VIA application instead for these keyboards. I'm unconvinced it is worth the effort of adding proper support for the new protocol.

GermanG, 2022-09-22:
I'm clueless about VIA, can you point me where to look for the protocol specs? (no promises, just want to figure how much effort it is)

xyzz, 2022-09-22:
I don't think there's a spec for the protocol, you can look at the source code - https://github.com/qmk/qmk_firmware/blob/master/quantum/via.c
as far as I can tell the changes are in https://github.com/qmk/qmk_firmware/commit/4efe6330c49f076b9c01d3f609dea0cd9f9b0214
however, the keycodes also changed in qmk between via's protocol 9 and 10 - specifically, qmk changed the TO() keycode macro, and I'm not sure what via plan is with regards to that, I think as of now it's just outright broken

GermanG, 2022-09-27:
Checking the commit you mention, looks like it is a change to manage the encoder knob, regretfully none of my keyboards have one so I can't check how it looks in the via web interface. (Not at home right now, but I'm going to check if the TO() messes my keyboard and edit this comment)