vial-kb/vial-gui #138: "Unsupported protocol version!" with 0.6 and default firmware

michaelcrusso, 2022-10-19:
I'm using a Nyquist Version 3, flashed through QMK Toolbox with the firmware package from the [Vial website](https://www.caniusevia.com/docs/download_firmware). Windows 10 x64. It seems to work perfectly through the webapp on Chrome, but as soon as I plug it in and open the desktop app, I get a warning saying "Unsupported protocol version!" exhorting me to get the latest version of Vial, which I already have. If I press OK, it goes on to show the next error below. It correctly identifies my keyboard as a Nyquist V3 in the dropdown box on top. I can see these same errors if I click About > Nyquist in the popup box.

```
File "main_window.py", line 270, in on_devices_updated
File "main_window.py", line 284, in on_device_selected
File "main_window.py", line 304, in rebuild
File "editor\keymap_editor.py", line 125, in rebuild
File "keycodes.py", line 870, in recreate_keyboard_keycodes

AttributeError: 'Keyboard' object has no attribute 'tap_dance_count'
```

I'm not a programmer so I'm really not sure what's going on here and haven't got the faintest idea how to debug it. I've tried installing various different versions of Python in case it's a dependency issue or something.

xyzz, 2022-10-19:
> with the firmware package from the [Vial website](https://www.caniusevia.com/docs/download_firmware)

this is the VIA website, not Vial website. The vial website is https://get.vial.today/. Since you flashed the VIA firmware to the board, you should be using the VIA application, which you can access from their website - https://usevia.app

closed 2022-10-19 by xyzz

t812206236, 2022-12-01:
None of qmk's firmware can use vial after qmk 's commit 4efe6330c49f076b9c01d3f609dea0cd9f9b0214

t812206236, 2022-12-01:
the via protocol version upgrade from 0x9 to 0xa

xyzz, 2022-12-02:
yes, you should use VIA if you compile qmk_firmware with VIA_ENABLE

wishinghand, 2023-02-07:
I'm having the exact same issue, but with a Sofle v2 RGB. I did download my firmware from [Vial's repo here](https://github.com/vial-kb/vial-qmk/tree/vial/keyboards/sofle/keymaps/vial). 

> None of qmk's firmware can use vial after qmk 's commit 4efe6330c49f076b9c01d3f609dea0cd9f9b0214

Does this mean I need to older QMK firmware releases?

> yes, you should use VIA if you compile qmk_firmware with VIA_ENABLE

When I deleted the line with `VIA_ENABLE` from my rules.mk, I couldn't even get my keyboard to be detected in Vial. If I keep that line, it will be detected but the GUI never loads any graphics to start configuring my keyboard.

xyzz, 2023-02-08:
You probably didn't build it correctly, make sure you're using the vial-qmk repo; the mention of VIA_ENABLE was for the case where VIA_ENABLE is present but VIAL_ENABLE isn't, either when compiling from the main QMK repo, or when not having the proper configuration enabled.

DerfOh, 2023-07-21:
I came here with the same issue using FW provided [here ](https://github.com/nullbitsco/docs/blob/main/firmware/firmware_flashing.md) and [here ](https://www.caniusevia.com/docs/download_firmware), am I to understand the solution is to recompile the source with a flag enabled? Perhaps the releases that are broken need to be pulled if that's the case.

xyzz, 2023-07-21:
> I came [here ](https://github.com/nullbitsco/docs/blob/main/firmware/firmware_flashing.md)with the same issue using FW provided [here](https://www.caniusevia.com/docs/download_firmware) and here

You've linked to firmware downloads for VIA - you need to use the VIA app with these.

DerfOh, 2023-07-21:
> > I came [here ](https://github.com/nullbitsco/docs/blob/main/firmware/firmware_flashing.md)with the same issue using FW provided [here](https://www.caniusevia.com/docs/download_firmware) and here
> 
> You've linked to firmware downloads for VIA - you need to use the VIA app with these.

Got it, thanks sorry for the confusion

xyzz, 2024-02-26:
There are no plans to support newer VIA protocols in Vial; if you'd like to utilize this extra Vial functionality you will need to install Vial-compatible firmware on your keyboard.

rbuzatu90, 2025-06-11:
I'm having the same issue. The keyboard gets detected but the config interfaces is not rendered. Is there a way to enable some kind of debug ? I'm a bit familiar with python and I would be able to debug this a bit.