vial-kb/vial-gui #145: Vial configurator tool does not allow to open a firmware that has more than 32 Macros

fghxu, 2022-11-21:
If someone uses the **vial-qmk** to build a firmware has the Macro_count more than 32,   then will not able to load the firmware with the vial configurator (GUI) tool.

For example:  if a firmware defines 48 macros. 
when trying to load the firmware with the vial (GUI) tool, there will be a pop up window says: ""expected array with 32 macros, got 48 macros"

Please open up the Macro_count  size limitation!   
Many keyboards have bigger EEPROM size now-a-days,  (mine has 2500+ memory for the macros, so I literally can have 100 macros.)

Nisker, 2022-11-21:
As of PR #135, there is no limit. 
You are probably running the latest release of Vial, which was released before the PR.
You can try downloading one of the artifacts that gets generated by github when a PR is tested, or run Vial from source.

fghxu, 2022-11-21:
Hi Nisker,
Thank you very much for the quick response and certify that the issue has been fixed.
I tried to follow the ReadMe.md (instruction) of the vial-gui git repo.  But unable to compile the python project. I am new to the python.

Can you please direct me the link that I can download the artifacts that is already generated by GitHub?

Thanks a lot
Frank

Nisker, 2022-11-21:
At the bottom of https://github.com/vial-kb/vial-gui/actions/runs/3131008074 you will find artifacts for windows and mac.
You need to be signed into github to download.

fghxu, 2022-11-22:
Thank you SO much, Nisker!
It’s working,  I am able to load my 108 macro firmware now in vial.

Thanks a bunch!
Frank

closed 2022-11-22 by xyzz