vial-kb/vial-gui #149: Unsupported VialRGB protocol

niclake, 2022-12-14:
Similar issue to https://github.com/vial-kb/vial-gui/issues/53

[image]

Attempting to flash a DZ60RGB V2.1.

Here's my `rules.mk`:

```

# MCU name
MCU = atmega32u4

# Bootloader selection
BOOTLOADER = lufa-ms
BOOTLOADER_SIZE = 6144

# Build Options
#   change yes to no to disable
#
BOOTMAGIC_ENABLE = yes      # Enable Bootmagic Lite
MOUSEKEY_ENABLE = yes          # Mouse keys
EXTRAKEY_ENABLE = yes          # Audio control and System control
CONSOLE_ENABLE = no            # Console for debug
COMMAND_ENABLE = no            # Commands for debug and configuration
NKRO_ENABLE = yes           # Enable N-Key Rollover
BACKLIGHT_ENABLE = no          # Enable keyboard backlight functionality
RGBLIGHT_ENABLE = no           # Enable keyboard RGB underglow
AUDIO_ENABLE = no              # Audio output
LTO_ENABLE = yes
VIA_ENABLE = yes
VIAL_ENABLE = yes
COMBO_ENABLE = yes
KEY_OVERRIDE_ENABLE = no
QMK_SETTINGS = no
WPM_ENABLE = no
MAGIC_ENABLE = no
MOUSEKEY_ENABLE = no
OLED_DRIVER_ENABLE = no
TAP_DANCE_ENABLE = no
VIALRGB_ENABLE = yes
RGB_MATRIX_ENABLE = yes
```

And my `config.h`:

```
/* SPDX-License-Identifier: GPL-2.0-or-later */

#pragma once

#define VIAL_KEYBOARD_UID {0x7E, 0x55, 0xFF, 0x5B, 0x6B, 0xC9, 0x0E, 0x8C}

#define VIAL_UNLOCK_COMBO_ROWS {0, 2, 3}
#define VIAL_UNLOCK_COMBO_COLS {0, 13, 0}

#define RGB_MATRIX_FRAMEBUFFER_EFFECTS
#define RGB_MATRIX_KEYPRESSES
```

soenkewindhausen, 2022-12-28:
fix: work through this https://get.vial.today/docs/lighting.html

ArijanJ, 2024-11-12:
This happened to me too, but my sofle_choc doesn't even have RBG. I had to set `"lighting": "none"` in vial.json, and looks like everything works now.