vial-kb/vial-gui #158: can you use more than 4 layers with VIAL?

csarmendariz, 2023-01-27:
Thanks everyone that contributed to this program, it's great!
Not really an issue, more of a question, is there an easy way to use more than 4 layers with VIAL?

Thanatermesis, 2023-01-27:
I was going to ask the same question :) / searching how to do that, same thing with the macros :thinking:  I want to create more than 15 but i dont want to write them directly from the firmware, better to configure all from VIAL

Nisker, 2023-01-27:
To get more layer you have to set it with 
`#define DYNAMIC_KEYMAP_LAYER_COUNT 8` in config.h (example with 8 layers).
Your keymap (in vial/keymap.c) should be modified to be the same size as your dynamic keymap count, so all keys have a predefined state.
As an example you could look at #315.

As for macros it is done with `#define DYNAMIC_KEYMAP_MACRO_COUNT 32` (example with 32 macros).

Thanatermesis, 2023-01-28:
Thanks for the info @Nisker !

Unfortunately I'm having trouble **switching from 4 to 5 layers**, it complains with:
```
Compiling: quantum/dynamic_keymap.c
quantum/dynamic_keymap.c:121:1: error: static assertion failed: "Dynamic keymaps are configured to use more EEPROM than is available."
 _Static_assert(DYNAMIC_KEYMAP_EEPROM_MAX_ADDR >= DYNAMIC_KEYMAP_MACRO_EEPROM_ADDR + 100, "Dynamic keymaps are configured to use more EEPROM than is available.");
```

So seems like the EPROM size is too small to have 5 layers :thinking: , then I found [a piece of code](https://github.com/vial-kb/vial-qmk/blob/vial/keyboards/system76/launch_1/config.h#L88-L108) that made it compile but I was not convinced that this was correct, after to research more I found that I simply need to increase the eprom size, unfortunately I cannot do it on my config.h but I need to do it from here [vial-qmk/platforms/eeprom.h#L56](https://github.com/vial-kb/vial-qmk/blob/vial/platforms/eeprom.h#L56), only changing 1024 to 2048 make it work, but now I wonder how much correct is that since its not meant to be changed (?) I can safely increase the size of the eprom? (my actual built firmware is 88% so I think I still have space), or is this a limitation of the chipset to be limited to 1024 kb?   

After a small search in google seems like indeed the atmega32u4 has only 1kb of eeprom, I wonder now how I can make it to have more layers :thinking: so if im not wrong, if I use the non-vial editor (so pure qmk) I can have more layers but on vial I have this limitation because vial uses eeprom to edit them? :thinking:

Nisker, 2023-01-28:
> can i safely increase the size of the eeprom?
No, The size of the eeprom depends on the chip since it's built-in, for the atmega32u4 it looks like you are correct with the 1KiB. It cant really be substituted with flash without doing some custom implementation (as far as i know). External eeproms does work but i would try turning off some of the features you dont use before soldering another chip on the pcb.

> After a small search in google seems like indeed the atmega32u4 has only 1kb of eeprom, I wonder now how I can make it to have more layers :thinking: 
Some general tips on reducing firmware, ram and eeprom size can be found here: https://get.vial.today/docs/firmware-size.html

> so if im not wrong, if I use the non-vial editor (so pure qmk) I can have more layers but on vial I have this limitation because vial uses eeprom to edit them? :thinking: 
Yes exactly, Vial uses the eeprom to store the keymap this way they can edited without touching the flash, whereas pure QMK just has this hard programmed into the flash.

Nisker, 2023-05-04:
> @Nisker I wonder how it works with rp pico (RP2040) then? It does not has EEPROM at all - so is it actually simulated via flash?

Yes it will work with RP2040 and you are right it's simulated in flash

> If so, do you know if it can be increased, since mine it appears has 16 MB of it?

[eeprom_driver.md](https://github.com/qmk/qmk_firmware/blob/master/docs/eeprom_driver.md#wear-leveling-rp2040-driver-configuration-idwear_leveling-rp2040-driver-configuration) has some documentation about the wear level options.
It seems to default to 4KiB of usable EEPROM while using 8KiB of actual flash.
If that is not enough, then you can change it in your `config.h` :
```c
/* EEPROM Driver Configuration */
#define WEAR_LEVELING_LOGICAL_SIZE 8192 /*8192 is just an example*/
#define WEAR_LEVELING_BACKING_SIZE 16384 // (WEAR_LEVELING_LOGICAL_SIZE * 2)
```

sandric, 2023-05-04:
@Nisker thanks a lot!

closed 2023-07-22 by xyzz