vial-kb/vial-gui #160: MacOS dmg file not downloading.

rushils20, 2023-02-23:
When download the Vial dmg on MacOS and attempting to open the application, Mac gives me an error stating that there's no guarantee that "this is not free of malware" and asks me to eject the disk image, without giving me any options to bypass it. When I try to use the Vial web version to change the color of my underglow light, I'm getting an Uncaught error, and compiler error in Chrome.

kbrock, 2023-02-24:
I just tested this out right now.

- Download Vial.dmg
- Open Vial0.6.dmg
- Open a new window, navigating to Applications
- Drag Vial from dmg window to Application Folder
- Right mouse click Vial.app choose open
- It gives some malware issue, I just hit escape or click cancel
- Right mouse click Vial.app choose open (again)
- It now gives the option to open anyway
- Profit

kurt-rhee, 2025-01-02:
this doesn't work for me unfortunately, after right clicking the second time i am not getting a different option to open vial anyways.  Using vial 0.7

xyzz, 2025-01-02:
the option has been moved to Settings -> Privacy & Security in newer versions of macOS, you will need to click Open Anyway on there