vial-kb/vial-gui #161: Save layout saves an empty file

kbrock, 2023-02-24:
I'm running vial 0.6 on a mac

This is getting very slick

I'm currently viewing a reviung 34 and reviung 41
it gives me the chance to "save current layout..."
The output file is empty for either keyboard.

- Are there ways to view an error log?
- Or ways to edit this in place to try and add debugging statements so that I can provide more information / or a bugfix?

(novice with python but advanced enough to have gotten a few PRs into QMK)

ObiWanKeoni, 2023-02-25:
Not super familiar with running this GUI locally for debugging, but I think [this](https://github.com/vial-kb/vial-gui/blob/fce9099004f5d53e55320131a0c797f07c7791b9/.github/workflows/main.yml#L43) should be easy enough to replicate.

(This is the GitHub action that builds for Mac - you should be able to clone this repo, make changes, then follow these steps with some minor adjustments to get some debugging going)

@xyzz may have a better method but this could get you started at least

kbrock, 2023-03-19:
Thnx. Curiously, is it working for you on a pc?

mbe24, 2023-10-10:
I'm on a PC / Win10, and for me it's only not working (VIA Firmware Protocol 11)