vial-kb/vial-gui #163: "LMZAError: Input format not supported by decoder" on startup

KiraGreifeneder, 2023-02-27:
File "main_window.py", line 270, in on_devices_updated
  File "main_window.py", line 274, in on_device_selected
  File "autorefresh\autorefresh.py", line 79, in select_device
  File "vial_device.py", line 51, in open
  File "protocol\keyboard_comm.py", line 79, in reload
  File "protocol\keyboard_comm.py", line 135, in reload_layout
  File "lzma.py", line 334, in decompress

LZMAError: Input format not supported by decoder

johnmcclain77, 2024-03-11:
Same error as well

LukeDRussell, 2024-05-03:
I'm getting this on Fedora 40 with Gnome. I wasn't getting the error while on Plasma.

baldur132, 2024-07-31:
Yes, this. It worked on my machine at home, but doesn't at on my work machine.

oXCookiXo, 2024-08-02:
getting the same error with web or desktop

mysonthebison, 2024-08-13:
Same error as well with a Kit Adam keyboard

jasonmaskell, 2024-09-10:
I'm having this issue also, with Ergodash keyboard and Helios controller. It works fine at home where I installed all the software to flash and build the firmware. But then on my work keyboard, both through the browser and the portable app, I get this error. Home is Windows 11, work is Windows 10.

rkristelijn, 2025-03-03:
I'm working on lily58, macbook m2 with Dell WD19T dock.

I can confirm:
- vial throws error if lily58 is connected via the dock
- directly on macbook it works with no problems

Would be great if this could be fixed <3

leigholiver, 2025-05-16:
WD19TB dock here, in case it helps anyone else i can connect my keyboard to Vial if I plug into the thunderbolt port on the dock rather than a usb-c port.