vial-kb/vial-gui #167: Unable to connect to devices on MacOS Ventura - M1 Mac

oscb, 2023-03-10:
Not sure when VIAL stopped working, but I have a belief it might be related to the OS Upgrade in my computer recently:

I keep getting this error when opening VIAL 0.6.0

```
  File "main_window.py", line 270, in on_devices_updated
  File "main_window.py", line 274, in on_device_selected
  File "autorefresh/autorefresh.py", line 79, in select_device
  File "vial_device.py", line 49, in open
  File "vial_device.py", line 26, in open

RuntimeError: unable to open the device
```

Then it shows up the window with my devices properly listed but no layouts.

Let me know if there's any data I can grab that might help!

SRLive, 2023-03-29:
same error for me but using mac os monterey on M1 Mac

SRLive, 2023-03-31:
Hi, I've been able to make it work by using a cal digit mini dock using the usb 2 port instead of connecting the my keyboard directly on a usb-c port

kirbysayshi, 2024-04-30:
I'm on a 2019 Macbook Pro (intel), and I encountered this error when my keyboard was plugged into my CalDigit TB4 dock.

Plugging the keyboard directly into the laptop fixed the problem, for anyone else that runs into this!