vial-kb/vial-gui #168: Vial does not load any keyboard layouts

KimKevin99, 2023-03-25:
Using a Sofle V2 from ergomech.store (JosefAdamcikSofle [VIA]) on Windows 10 Lenovo laptop raises this error:   

File "main_window.py", line 270, in on_devices_updated
  File "main_window.py", line 284, in on_device_selected
  File "main_window.py", line 304, in rebuild
  File "editor\keymap_editor.py", line 125, in rebuild
  File "keycodes.py", line 870, in recreate_keyboard_keycodes

AttributeError: 'Keyboard' object has no attribute 'tap_dance_count

after a dialogue that says "Unsupported protocol version! Please download latest Vial from https://get.vial.today/ (despite running v0.6)

xyzz, 2023-03-25:
New version of VIA firmware isn't supported in Vial, use the VIA application instead

closed 2023-03-25 by xyzz

X4ndras, 2025-05-07:
via also does not work