vial-kb/vial-gui #171: Desktop vial cannot recognize my keyboard, but web vial can

HaoboGu, 2023-04-08:
Hello, I'm porting vial to my keyboard. I generated my keyboard definition using [vial_generate_definition.py](https://github.com/vial-kb/vial-qmk/blob/vial/util/vial_generate_definition.py), and my keyboard can be automatically recognized by web vial(vial.rocks). However, when I use desktop vial v0.6, it cannot detect my keyboard automatically. I have to sideload vial.json to use it. I guess I did miss something when porting vial, does anyone has any ideas about this? thanks!

xyzz, 2023-04-08:
you're not supposed to run that script, it's executed during the build. Most likely it's an issue of missing usb serial number, but I can't tell for sure without looking at your keyboard source code.

HaoboGu, 2023-04-09:
I set usb serial number for my keyboard to "000000000001", I sniffed usb communication and after I opened desktop vial, there are 6 requests to my keyboard to request 3 string descriptor.

Here is my keyboards response:

1st: usb serial number
[image]

2nd: vendor
[image]

3rd: product
[image]

Each request repeated once. No other requests.

@xyzz  Can you see any error of those responses?

xyzz, 2023-04-09:
If you change the serial number yourself, it must include the Vial serial magic (`vial:f64c2b3c`) somewhere in order to be detected.

HaoboGu, 2023-04-09:
Changing servial number fixed the issue, thank you very much @xyzz !

closed 2023-04-09 by HaoboGu

dostoievsky, 2023-06-14:
I am having this same issue, but after web loads the page, it says the same thing as desktop client.
But it is funny, because it detects when I start gui and then disappear.
I am using YMDK air 40. Can you guys help me?