vial-kb/vial-gui #172: Will TL_LOWR and TL_UPPR be added?

doesntfazer, 2023-04-11:
It looks like the way that Tri-Layer-State is handled has changed. I was wondering when/if this feature will be available now that it's as simple as adding `TRI_LAYER_ENABLE = yes` to the `rules.mk`

https://docs.qmk.fm/#/feature_tri_layer?id=tri-layers

Looks like there's a couple more things you can do like define which layers are tri-layer in the config.h.

xyzz, 2023-04-12:
The feature has been available in vial since the release under these two buttons:
[image]
if the question is whether it's gonna switch to the new tri-state layer implementation then the answer is yes

closed 2023-04-12 by xyzz

doesntfazer, 2023-04-12:
Thank you for the heads up! I was actually unaware of this.