vial-kb/vial-gui #184: RGB/LED effects defined in `info.json` instead of `config.h` do now show up in the Lighting tab over in the Vial GUI.

adophoxia, 2023-05-26:
As title. A user over in the Keychron server had the same issue with their Keychron Q0 Plus flashed with vial firmware from `vial-kb/vial-qmk` where 22 effects defined in `q0/rev_0131/info.json` did not show up for them. Even more apparent testing this for myself that none the effects aren't enabled in the firmware itself, with only `solid color` being the 1 effect enabled. Seems like until the GUI is updated to take in these effects in `info.json` or if the `merge-2023-03-12` branch has anything related to this, then having said effects back inside `config.h` would be a fix.

xyzz, 2023-06-25:
Defining effects in the json isn't supported in the current `vial` branch, need to wait for the `merge-2023-06-03` to be merged, or define them in `config.h` for now.

xyzz, 2023-07-22:
Should be resolved now

closed 2023-07-22 by xyzz