vial-kb/vial-gui #189: User defined custom keycode limited to 16

Alectardy98, 2023-07-20:
When trying to make more then 16 custom keycodes, using QK_KB_0, the GUI will not open and gives an error like shown in vial 0.7
[image]

Alectardy98, 2023-07-21:
this can be fixed by changing the limits defined in src/main/python/keycodes/keycodes_v6.py at line 578

Alectardy98, 2023-07-21:
[image]
 
 please make this change, it will solve a lot of my issues. Here is a photo of it working currently i have tested with it set to 50

xyzz, 2023-07-22:
Fixed in 64e725971ed7ba143cf188f5985e6e3eed776c3e

closed 2023-07-22 by xyzz

Alectardy98, 2023-07-22:
thank you, when will this be pushed into release?

Alectardy98, 2024-04-11:
[image]

Has the fix been implemented yet?