vial-kb/vial-gui #192: Lighting Controls for Underglow Not Working

a33bis, 2023-08-03:
In the process of adding back support for the infinity87 rev1 standard ([vial-qmk#514](https://github.com/vial-kb/vial-qmk/pull/514)) that was removed in the last merge I realized it is not possible to control the underglow from the "Lighting" tab anymore.

By default the underglow of the board is set to solid red with max brightness but by default the tab display "All Off", 0 brightness and black. I can select different effects but it does nothing, I can click and choose a different color but right after clicking "Ok" it stays on black, and lastly I can not interact at all with the brightness slider.
[image]
Note that using the RGB control keycodes everything works fine (the "Lighting" tab  stays however as described)
Also as I said I already ported this specific keyboard and it worked fine back then (3months ago) however it seems the keyboard maintainer changed a lot in the overall organization, notably moving partially to data driven configuration which I suspect is causing the issue since it is essentially the only difference from before.

a33bis, 2023-08-04:
Tested with a different pcb (Galatea Rev2) same issue.

a33bis, 2023-08-06:
Backlight controls also do not work, I can interact with the slider unlike with underglow but it has no effect, must use keycodes.

xyzz, 2023-08-12:
Fixed in https://github.com/vial-kb/vial-qmk/commit/ca3a3d8030f2d3395e84efcc64ac943e99ca9fbc

closed 2023-08-12 by xyzz