vial-kb/vial-gui #198: Unable to resolve qmk_id=LT16(kc)

dalton-elmo, 2023-08-12:
The first time I tried to modify my keyboard layout everything went well, but after a few days I got this error:

``` 
  File "main_window.py", line 279, in on_devices_updated
  File "main_window.py", line 283, in on_device_selected
  File "autorefresh/autorefresh.py", line 79, in select_device
  File "vial_device.py", line 51, in open
  File "protocol/keyboard_comm.py", line 90, in reload
  File "keycodes/keycodes.py", line 1062, in recreate_keyboard_keycodes
  File "keycodes/keycodes.py", line 960, in recreate_keycodes
  File "keycodes/keycodes.py", line 133, in deserialize
  File "keycodes/keycodes.py", line 157, in resolve

RuntimeError: unable to resolve qmk_id=LT16(kc)
```

I used Vial 0.7 appimage version on Arch Linux, but web version also doesn't work.

Some hardware information:
```
Manufacturer: ergohaven
Product: k:02
VID: FEED
PID: 9918
Device: b'/dev/hidraw3'

VIA protocol: 9
Vial protocol: 5
Vial keyboard ID: 240D36AA983F003

Macro entries: 32
Macro memory: 1021 bytes
Macro delays: yes
Complex (2-byte) macro keycodes: yes

Tap Dance entries: 32
Combo entries: 32
Key Override entries: 32

QMK Settings: yes
```

xyzz, 2023-08-12:
Does that still happen on the nightly version from https://github.com/vial-kb/vial-gui/actions/runs/5839488107 ?

dalton-elmo, 2023-08-12:
No, there are no problems on the nightly version. Thanks.

closed 2023-08-12 by dalton-elmo