vial-kb/vial-gui #209: [request] Support for keychron k13

appetrosyan, 2023-10-18:
Trying to use vial with a keychron k13, I have to load the following JSON, which results in a regular sideload showing up as the correct keyboard, however, trying to switch to it from the drop-down menu results in  

[image]

appetrosyan, 2023-10-24:
Suppose I'm a decent C-programmer, what can I do?

appetrosyan, 2023-10-25:
In that case, let's leave this issue open, I'll post updates on my success here. I will have some spare time in December so I'd expect that it'd be a while.

appetrosyan, 2023-12-04:
The Pro variant. The regular doesn't even advertise QMK.

jckytf, 2024-02-24:
There's a Max variant too. Hope it gets support also.

appetrosyan, 2024-02-24:
So I got in touch with Keychron. 

They don't want to even admit that the keyboard has any issues with QMK. From what I can understand if I don't do it myself, it won't get done at all. 

I don't have a lot of experience with Firmware, but I'm a decent systems programmer, maybe I'd be able to create a keymap.

appetrosyan, 2024-02-24:
On that note, Keychron k13 doesn't support a Via v3 firmware either. It requires me to specifically choose the V2 protocol.... I'm not a huge fan of how it works, but I managed to create some customisation with the `caniusevia.app` maybe I can port that to vial and not have to use a browser to customise my hardware.

adophoxia, 2024-02-24:
> They don't want to even admit that the keyboard has any issues with QMK. From what I can understand if I don't do it myself, it won't get done at all.

There is no issue with this board regarding QMK. You are right about the other half of this paragraph, considering the Pro/Max variants of these boards aren't in mainline QMK. 

> On that note, Keychron k13 doesn't support a Via v3 firmware either. It requires me to specifically choose the V2 protocol.... I'm not a huge fan of how it works, but I managed to create some [...]

*K13 Pro, and how is the first half related to Vial?

appetrosyan, 2024-03-05:
> There is no issue with this board regarding QMK.

The board is not to be merged into mainline, and nobody seems to be doing any work in that direction. 

The issue is that they claim full QMK support, when in reality all they have is a small folder in the `bluetooth_playground` that needs a lot of work. Flashing that firmware no longer works on my machine. It needs to be updated. And nobody seems to be interested. 

> *K13 Pro, and how is the first half related to Vial?

It isn't exactly. Vial seems to not be using Via, but I only realised it once replying to your message. 

So on that note @adophoxia, where can I learn how to do this? I'm a decent programmer, but I don't have the domain knowledge to work with firmware. If it were easy, I'm sure Keychron would have done this by now. So where can I "git gud"?

appetrosyan, 2024-06-04:
Can I ask you to at least close it as not planned?

xyzz, 2024-06-04:
There you go; in general Vial considers board addition/support as responsibility of the vendor.

closed 2024-06-04 by xyzz