vial-kb/vial-gui #219: Keyboard detection doesn't work

lsahnicne, 2024-01-01:
I have a Keychron S1 keyboard.
I'm using Pop!_OS 22.04 LTS.
I've downloaded Vial 0.7.1 as an AppImage file.
I've followed the instructions at https://get.vial.today/manual/linux-udev.html and created a file '99-vial.rules' at /etc/udev/ryles.d/ with the following text inside:
```
# Keychron S1
KERNEL=="hidraw*", SUBSYSTEM=="hidraw", ATTRS{idVendor}=="3434", ATTRS{idProduct}=="0410", MODE="0660", GROUP="users", TAG+="uaccess", TAG+="udev-acl"
```
After that I've tried both rebooting my PC and manually executing the following two commands iin terminal:
```
sudo udevadm control --reload-rules
sudo udevadm trigger
```
However, when I launch Vial AppImage, it still says 'No devices detected'.
Pressing File-Download VIA definitions doesn't help.

xyzz, 2024-01-02:
is the keyboard flashed with Vial firmware?

lsahnicne, 2024-01-02:
> is the keyboard flashed with Vial firmware?

Missed that step, my bad.

closed 2024-01-02 by lsahnicne