vial-kb/vial-gui #220: Bug with key overrides + modifier behaviour

esn89, 2024-01-02:
I am trying to emulate the "RAlt + Backspace" behaviour which currently works.

It backspaces an entire word and lands the cursor in front of the word with a space:
(I use `][` to denote the cursor position)
Before:

`foo bar][` 

I press "RAlt +Backspace"

After:

`foo ][`

Now, because of muscle memory, I am more used to hitting "LCtrl + Backspace" for this exact behaviour, so I set up a Key Overrides on my backspace to behave like "RAlt + Backspace" when I hit "Regular Backspace with my "LCtrl".  I have it set up like so:

[image]

My LCtrl is located where my Caps Lock is as a Mod Tap:

[image]

The UNEXPECTED behaviour of "LCtrl + Backspace" is that the cursor lands in front of the word, without a space:

`foo bar][` 

I press "RAlt +Backspace"

After:

`foo][`

Am I missing something?  Is this a bug or a misconfiguration?

xyzz, 2024-01-02:
You need to add Ctrl to the Suppressed mods, otherwise what you're actually entering is Ctrl+Alt+Backspace.

closed 2024-01-02 by xyzz

esn89, 2024-01-02:
@xyzz 

Amazing, thanks for that quick fix, it works.  Would you mind explaining though?  I had thought that by "supressing" something, it'd make it not work.  How did adding "LCtrl" as a "Supressed mods" make this work?

xyzz, 2024-01-02:
the key overrides documentation is available in qmk at https://github.com/qmk/qmk_firmware/blob/master/docs/feature_key_overrides.md, in short, a suppressed mod key will release the corresponding modifier at the same time that it's overriding the key