vial-kb/vial-gui #226: newer version or protocal of vial not able to read or configure keyboard layout.

CameronWeller, 2024-02-19:


xyzz, 2024-02-20:
> the newer compiled keyboard is unable to work with older versions of the vial GUI, I get an error specifying a protocol change.

You need to use the newest available version of Vial, there is no goal of forwards compatibility (only backwards compatibility)

closed 2024-02-20 by xyzz

CameronWeller, 2024-02-20:
> > the newer compiled keyboard is unable to work with older versions of the vial GUI, I get an error specifying a protocol change.
> 
> You need to use the newest available version of Vial, there is no goal of forwards compatibility (only backwards compatibility)
I am not saying that was the issue. Just wanted to specify that I tried ;(

xyzz, 2024-02-20:
If you get a protocol version error with the latest version of Vial, then it means you're trying to use it with a VIA keyboard which is no longer supported. You should be either 1) using the VIA application with that keyboard, or 2) flash the Vial firmware on the keyboard following manufacturers instructions

CameronWeller, 2024-02-22:
(Before I continue I just want to say I appreciate your help and effort making this project)

But yeah this is a custom keyboard that wrote the config for and am flashing myself. I only ever intended to use this keyboard project of mine with vial support, I have not built any via support into the project except what was necessary for getting vial up and running.