vial-kb/vial-gui #237: KVM - Vial Native - Windows 11 not detecting Keyboard

fmunozs, 2024-04-14:
Hello,

I am using Windows 11 and a KVM (AIMOS HDMI KVM Switch Box, 4 in 1 Out UHD 4K@60Hz, USB Switch Selector with 4 USB2.0 Hub) and the VIAL Windows app is not able to detect the keyboard.

Web based configurator https://vial.rocks works okay and if I plug the KB directly to the desktop computer and not through the KVM it also works on the same machine.

fmunozs, 2024-04-15:
Did some additional debugging with a local copy of vial-gui, seems like the KVM actually changes the serial of the keyboard (for reasons unknown)

Using KVM:
VID=FEED, PID=B155, serial=vial:f64c2b31

Not using the KVM:
VID=FEED, PID=B155, serial=vial:f64c2b3c

Since the serial is changed, it is not detected as a VIAL KB.

fmunozs, 2024-04-15:
Confirmed by changing the serial on the util.py to "vial:" allows me to use vial-gui through the KVM. Not sure if any one else could be affected by a problem like this though

Tested on Windows and on a Ubuntu machine (also required to update the udev rules).

[image]

jasonzjc666, 2024-04-26:
Similar issue here - I just upgraded from 0.4.2 to 0.7, and vial (both web and windows native) stopped recognizing my keyboard which is conneted via a USB hub. Everything is back to normal if I connect the keyboard directy to the PC.

sehrgut, 2025-11-14:
Since the KVM is rewriting the USB stack information, this probably needs to be addressed with the KVM manufacturer. While I couldn't find anything in the manual online for your KVM, it's possibly some kind of workaround for determining which connected computer is sending status LED packets to the keyboard. A kind of primitive "USB NAT" as it were.

Does it show the same rewritten serial on each connected computer, or different ones?

fmunozs, 2025-11-14:
The problem is that the KVM changes the serial and the detection method for vial-gui requires it to identify the keyboard. As I mentioned, if I change the serial from the vial-kvm code, I was able to edit configuration without issues on my board through the KVM, also the detection code on the web editor is different.

bendavis987, 2025-12-05:
I confirmed this is the issue I'm seeing as well. I assumed my AV Access B30 dock just didn't support the VIAL keyboard.

1. I confirmed my keyboard is detected by VIAL on any USB port besides the KVM designated keyboard port.
2. I confirmed my keyboard is detected in the web (vial.rocks) on the KVM designated keyboard port and on other USB ports.
3. I confirmed my KVM does change the device serial number as described by @fmunozs.
- vial:f64c2b3c
- vial:f64c2b34
4. I wasn't able to view kde plasma usb devices from my laptop. It only lists thunderbolt. The laptop did not recognize the VIAL keyboard on any port in VIAL GUI.
5. Vial.rocks was able to find it on both the KVM designated port and the USB port on the dock. These both were not able to establish a connection over the thunderbolt connection.
6. Vial.rocks was not able to connect over the laptop's USB port either. Seems my laptop has another issue that I need to look into.