vial-kb/vial-gui #255: RuntimeError: failed to communicate with the device

KashmiriKanna, 2024-07-23:
File "main_window.py", line 279, in on_devices_updated File "main_window.py", line 283, in on_device_selected File "autorefresh\autorefresh.py", line 79, in select_device File "vial_device.py", line 51, in open
File "protocoN\keyboard_comm.py", line 79, in reload File "protocoN\keyboard_comm.py", line 115, in reload_layout File "protocoN\keyboard_comm.py", line 105, in reload_via_protocol File "util.py", line 66, in hid_send
RuntimeError: failed to communicate with the device

joshu-hua, 2024-10-24:
I'm also getting this issue with my Elora from splitkb

keebicrunk, 2025-01-17:
I am also getting this issue with all keyboards I plug into downloaded vial. I updated to 0.7.1 beforehand but it's not reading any device.