vial-kb/vial-gui #264: `error: argument out of range` when loading layout

vdawg-git, 2024-09-20:
Whenever I try to load a layout for my Piantor Pro I get this error:
```
  File "main_window.py", line 245, in on_layout_load
  File "editor/keymap_editor.py", line 144, in restore_layout
  File "protocol/keyboard_comm.py", line 416, in restore_layout
  File "protocol/keyboard_comm.py", line 318, in set_key

error: argument out of range
```

I tried to load those two layouts (one of them is not for the Piantor though):
[piantor36_miryoku.vil.json](https://github.com/user-attachments/files/17080955/piantor36_miryoku.vil.json)

[cantor-miryoku-mac.vil.json](https://github.com/user-attachments/files/17080963/cantor-miryoku-mac.vil.json)

They were both `.vil` originally but Github wants the `.json` file extension.

vdawg-git, 2024-09-21:
Seems to an issue with the files I tried to load. 
Not sure what the problem was. 

I leave it open as a nicer error message would be nice :D

dwrth, 2025-07-08:
I have the same issue with my Corne v4, did you manage to work around this issue?