vial-kb/vial-gui #274: @ in Text macro

Domme0506, 2024-11-08:
Hello ,

I would like to create a specific macro in which my email address should be written. But if I use an @ symbol, instead of the @ symbol there is only a "

 Is it possible to implement this differently?

Kind Regards

Nisker, 2024-11-08:
The @ symbol expands to the keycode LSFT(KC_2) because on a US keyboard Shift + 2 makes an @  
Suppose you used a Scandinavian keyboard this you would get a " since Shift + 2 is " whereas AltGr + 2 is @.  
So on Scandinavian keyboards you would use the RALT(KC_2) keycode.

Domme0506, 2024-11-08:
and how can i use this combo in the macro editor ?

[image]

peterjc, 2025-03-02:
Try three steps,
1. text for the first part of the email
2. send tap key RALT(KC_2)
3. text for the second part of the email