vial-kb/vial-gui #296: Tap Dance Issues with Caps Lock

quincyp, 2025-01-19:
[image]

Tap dance does not seem to like tapping caps lock. Caps lock will not work on tap, or on double tap. It will however work if I swapped caps lock to on hold and the other key to either of the taps. 

Additional info: I found that after some testing, on tap/double tap, with caps lock set it will disable caps lock, but it will not enable caps lock. 

Please update as it makes sense to have it work on tap not hold for this specific use case

danilomourelle, 2025-03-31:
If I may, I have this exact setup on my sofle and it works well to enable and disable caps. Try swap switches update tapping term to 300.

quincyp, 2025-03-31:
thanks for the reply, what do you mean by enable and disable caps and swap switches?
perhaps a screenshot would be quicker than text for this case?

So I set my Tap Dance tab to this
[image]

And on my keymap the key is set to this
[image]

I've tried the 300 tapping term along with shorter/longer durations and it doesn't seem to change

danilomourelle, 2025-05-08:
Hey man, sorry for late response. I mean swap the swiches to see if it is a hardware problem. But since the last response, I found out this key. I it's much better for this shift/caps config the tap dance. You should give it a try

[image]

quincyp, 2025-05-08:
all good, notifications get lost all the time, that solution is perfect! I previously tried Lsft (kc) and it didn't work. But I noticed your image is Lsft_T (kc) instead which works for this exact use case!

Sidenote: do you know what Lsft (kc) does instead? Didn't find the answer on a quic search trying to see if any of these other keys under "Quantum" could be useful

quincyp, 2025-05-08:
Update: Lsft (kc) appears to just be for a shfited key. IE if you wanted a key to be just ! you would put 1 there or just ? you would put / there. The _T are for tap vs hold shortcuts, thanks again!

closed 2025-05-08 by quincyp