vial-kb/vial-gui #302: [Feature request] Add Caps Word as a Vial feature.

getreuer, 2025-02-09:
## Overview

Caps Word has been available as a core [QMK feature](https://docs.qmk.fm/features/caps_word) since 2022 and available [in Oryx](https://blog.zsa.io/2206-caps-word/) since 2023. It is a popular feature, especially for home row mods users to limit the need to hold Shift.

It would be great for Vial's users to expose Caps Word in the Quantum tab. I am interested to contribute PRs to making that happen, but I need help on clarifying the right approach. That's what I hope to discuss in this issue. Or if not here, let me know where better to discuss this.

## Outline

Here is an outline of how I think a "Caps Word" button could be added in Vial:

1. In vial-kb/vial-gui, define the Caps Word keycode `"CW_TOGG"` (= [`0x7C73`](https://github.com/qmk/qmk_firmware/blob/2699e2f7c826b0288c3dddcd86a97811726cf6a4/quantum/keycodes.h#L756) in Vial's v6 keycodes) in src/main/python/keycodes/.

2. Add `"CW_TOGG"` to `KEYCODES_QUANTUM` in src/main/python/keycodes/keycodes.py. Then (?), Caps Word will appear as an available button under Vial's Quantum tab.

3. In vial-kb/vial-qmk, enable Caps Word in the keyboard firmware. This would (?) be done independently for each keyboard that opts to do so by adding "`CAPS_WORD_ENABLE = yes`" in \<*keyboard*\>/keymaps/vial/rules.mk.

4. In vial-kb/vial-kb.github.io, document the Caps Word feature (perhaps?).

## Questions

Keycode `CW_TOGG` would of course only work where the firmware was built with `CAPS_WORD_ENABLE = yes`. How to handle this consistently between the keyboard and GUI? Does Vial enable certain features by default for all keyboards, rather than what I said in (3) above with enabling per keyboard? Or is there a protocol for the keyboard to communicate to the GUI which features it has been built with so that disabled keycodes can be hidden, anything like this? 

Is it relevant to update only the v6 keycodes? Or do people still build Vial firmware using the v5 keycodes? `CW_TOGG` did exist prior to QMK's refactoring to data-defined keycodes, which I think is what Vial's v5 keycodes refer to. 

Does the above outline need anything further? Let me know what else I'm missing.

Thanks for your help!

xyzz, 2025-02-09:
Thanks for filing this issue, I really appreciate the amount of research you've done already here and here's to answer some questions and comment on the other pieces:

> In vial-kb/vial-qmk, enable Caps Word in the keyboard firmware. This would (?) be done independently for each keyboard that opts to do so by adding "CAPS_WORD_ENABLE = yes" in <keyboard>/keymaps/vial/rules.mk.

Generally Vial policy is to enable every feature by default and then it can be disabled for keyboards in the repo that run out of memory

> In vial-kb/vial-kb.github.io, document the Caps Word feature (perhaps?).

That's not really required, people can refer to QMK docs.

> Or is there a protocol for the keyboard to communicate to the GUI which features it has been built with so that disabled keycodes can be hidden, anything like this?

This isn't in place yet, but it really should be - I'm thinking we should stuff bits in the `dynamic_vial_get_number_of_entries` message since it's semantically the same thing even though it's talking about "number of entries" https://github.com/vial-kb/vial-qmk/blob/1c129915f059c004155a3e69df9068089ca753bb/quantum/vial.c#L222 e.g. starting from `msg[3]` here you can put the least significant bit for whether CAPS_WORD is enabled and we can add more on top in the future.

> Is it relevant to update only the v6 keycodes? Or do people still build Vial firmware using the v5 keycodes? CW_TOGG did exist prior to QMK's refactoring to data-defined keycodes, which I think is what Vial's v5 keycodes refer to.

There's no need to put the new keycode into v5 in this case, but keep in mind that v5 firmware should still work (i.e. vial gui shouldn't throw exceptions when a v5 keyboard is connected).

There's unfortunately a lot of nastiness in the keycode stuff as keycodes were implemented as a global originally but now we of course understand they belong to the keyboard so there's a ton of hacks that attempt to dynamically keep these global arrays up to date based on which keyboard version is connected & what features it has (like number of layers or tapdances). Ideally eventually it should be refactored, but it's of course not a burden I'd put on contributors so if you can get that feature in place in a way that doesn't make the rest of the gui keycode mess collapse on itself then that's just fine.

getreuer, 2025-06-18:
@xyzz thank you so much for the detailed and helpful response!

I've been tinkering on this and have integrations working to add Caps Word and also (while I'm at it) Layer Lock to Vial's firmware + GUI.

I plan to open a pair of associated PRs for these changes, one on vial-gui and one on vial-qmk. Let me know if there's a better way about how to approach review. Thanks!

xyzz, 2025-06-23:
Basic support for both flow tap and chordal hold is available in the next upcoming merge branch https://github.com/vial-kb/vial-qmk/commits/merge-2025-06-21/ and the corresponding GUI changes are in https://github.com/vial-kb/vial-gui/commits/new-tap-holds/

regarding configuring them per key it would be a much bigger change and I'd need to think how best to approach it, if it requires a new widget it should be designed in a way that could be reused for other settings too (since the eventual goal is to use the existing qmk settings framework to let keyboards define custom settings following the existing json syntax)

xyzz, 2025-06-24:
> Communication: Define a new "set_per_key_setting" message type of a one-byte qsid followed by up to 31 bytes for the new data block. 31 bytes is enough for 248 keys, is that enough to support all real keyboards? If not, the data block can be transferred in chunks like the protocol used for the keymap buffer.

I think that should be ok if optimally packed; with VIA matrix tester we discovered that the non-optimal packing done with `((MATRIX_COLS / 8 + 1) * MATRIX_ROWS <= 28)` is actually not enough for some real world keyboards.

As for making it generic enough, a qsid / qmk_setting entry with the meaning of on/off toggle for each individual key sounds okay to me and hopefully could be useful for other features.

> that look doable in Vial are:

Yes, for some of these however the concern would be as I described above that there's no good way for us to communicate the feature enable for keycodes; conversely for qmk settings it is easier since we can only define qsid if it is enabled in the firmware. So for some of these (caps word, layer lock, repeat key) you run into the issue that it ideally needs to be dynamically hidden based off keyboard compile time features, and that runs into the horrible mess that is the current GUI keycode implementation.

getreuer, 2025-06-24:
Thanks for the quick response!

> Yes, for some of these however the concern would be as I described above that there's no good way for us to communicate the feature enable for keycodes

Ah, right, I ran into that while tinkering an integration for Caps Word and Layer Lock. Here is how I approached it. It's not particularly clever, but does the job:

1. On the firmware side, as you suggested, use bits of `msg[3]` in `dynamic_vial_get_number_of_entries` to indicate when optional features are supported in firmware (e.g. bit 0 ⇒ Caps Word, bit 1 ⇒ Layer Lock, bit 2 ⇒ Repeat Key, ...).

2. On the GUI side in `protocol/dynamic.py`, do

    ```py
    def reload_dynamic(self):
        self.supported_features = set()  # Added.
    
        if self.vial_protocol < VIAL_PROTOCOL_DYNAMIC:
            self.tap_dance_count = 0
            self.tap_dance_entries = []
            self.combo_count = 0
            self.combo_entries = []
            self.key_override_count = 0
            self.key_override_entries = []
            return
        data = self.usb_send(self.dev, struct.pack("BBB", CMD_VIA_VIAL_PREFIX, CMD_VIAL_DYNAMIC_ENTRY_OP,
                                                   DYNAMIC_VIAL_GET_NUMBER_OF_ENTRIES), retries=20)
        self.tap_dance_count = data[0]
        self.combo_count = data[1]
        self.key_override_count = data[2]
    
        # Bits of data[3] indicate optionally supported features.
        for bit_index, feature in [
            (0, "caps_word"),
            (1, "layer_lock"),
            (2, "repeat_key"),
            # Add more feature bits as needed...
        ]:
          if data[3] & (1 << bit_index):
            self.supported_features.add(feature)
    ```

3. In `keycodes.py`, I added a "`requires_feature`" field to `Keycode`, defaulting to `None`, that may be set to the name of a require feature for that keycode. Example:

   ```py 
   K("QK_CAPS_WORD_TOGGLE", "Caps\nWord", "Capitalizes until end of current word", 
     alias=["CW_TOGG"], requires_feature="caps_word"),
   ```
4. Also in `keycodes.py`, finally, unsupported keycodes are suppressed from being shown by removing them at the end of the `recreate_keyboard_keycodes()` function like this:

   ```py
   # Keep only keycodes where .requires_feature is in keyboard.supported_features.
   def _keyboard_supports_keycode(keycode):
       if keycode.requires_feature is not None:
           return keycode.requires_feature in keyboard.supported_features
       return True

   KEYCODES[:] = list(filter(_keyboard_supports_keycode, KEYCODES))
   ```

Once this is set up, what's nice about this approach is that further feature-dependent keycodes are easy to add. A new feature and its keycodes would amount to updating `protocol/dynamic.py` to check for the feature bit and defining the keycodes with the `requires_feature` field.

WDYT? Happy to make adjustments.

getreuer, 2025-06-25:
Great! I just opened a pair of PRs targeting the upcoming branches for PDF(layer), Caps Word, and Layer Lock:

* Firmware: https://github.com/vial-kb/vial-qmk/pull/902
* GUI: https://github.com/vial-kb/vial-gui/pull/332

getreuer, 2025-06-29:
@xyzz thank you for the reviews! Now that I'm getting a bit more familiar with Vial's code, I'd like to propose a somewhat more ambitious change, adding Repeat Key and Alt Repeat Key with a dynamic configuration to customize what Alt Repeat Key does:

* Firmware: https://github.com/vial-kb/vial-qmk/pull/906
* GUI: https://github.com/vial-kb/vial-gui/pull/334

Screenshot:

[image]

xyzz, 2025-07-03:
In this case, I manually disabled all 3 features for keyboards which failed the Vial keymap build in the latest action https://github.com/vial-kb/vial-qmk/actions/runs/15952449843/job/44994284848?pr=906. Note that in your link for "github action build log " you're actually linking to an even older commit, specifically it's the "qmk_settings: add flow tap" one.

Since there were so many failing boards (~170) and I was going through these manually, I just turned off the latest 3 optional features, but of course a smarter approach could be used based off diffing the caps word build log to see which boards can keep it (https://github.com/vial-kb/vial-qmk/actions/runs/15948513692/job/44985508891)

Vial policy has been that it should be up to the keyboard vendor to supply the proper firmware builds, at which point they can customize the features needed etc - whereas automated builds in vial-qmk mostly exist to ensure that newer firmware or Vial changes do not break real world usages in keyboards (this is also why Vial intentionally does not provide firmware downloads for any keyboards); of course in practice it's a mixed bag. Ultimately, if you'd like to, you're definitely welcome to open a PR to reenable Caps Word on some of these keyboards and I'll merge it no problem if it can pass the github CI.

reinier, 2025-07-11:
I'm not sure if a post like this (where nothing is added to the the thread accept for some positive vibes) is allowed but I want to say thank you @getreuer for your work on this. I was looking for a way to use layer lock and caps word in Vial for my new keyboard (a Halcyon Ferris) and was disappointed when I couldn't find it anywhere. I'm looking forward to this update!

getreuer, 2025-07-13:
@reinier thanks for the note, much appreciated! Enjoy the new features 🥳