vial-kb/vial-gui #306: KVM: Keyboard detected as VIA device when plugged into KVM HID port

rjorgenson, 2025-02-26:
I have a KPRepublic XD75 that was previously flashed to VIA. I have flashed it to Vial and everything was working great until I plugged the keyboard back in to the HID port for keyboard control of the KVM switcher. My layer keys were sending control and shift keys instead. I checked the map in Vial and there were weird keycodes on the layer keys, I set them back to the correct keycode but it didn't change anything. After a while of staring at this I noticed that there were almost no configuration options, just keymap, macros, and lighting and the keyboard selector was listing the device as `xiudi XD75 (VIA)`, plugged it back into a real usb port (still on the same KVM device and sure enough the device is reported as `xidui XD75` and all of the configuration options are back. These screenshots are of the same keyboard plugged into different ports on my KVM, one of them an HID port.

Is there something special I need to do to enable support for this keyboard on the HID port? The VIA firmware worked through this port without issue for years. #237 mentions a different serial number, but I'm not sure how to tell what that is reporting or if it is different.

[image]

[image]

rjorgenson, 2025-02-26:
I did figure out how to find the serial information, sure enough it is different between the two USB ports.

HID port:
[image]

USB3 Port:

[image]

I have not yet figured out how to change the code in Vial.app that worked as a workaround in #237 yet, but going to give that a try once I can.

xyzz, 2025-02-26:
if the kvm strips the serial number or modifies it then the desktop app won't probe it as a vial keyboard; you can try the web version since it uses a slightly different detection algorithm

rjorgenson, 2025-02-27:
I gave vial web a try and not only did it detect the keyboard properly it also didn't change the behavior of the keyboard when using the HID port. After confirming that worked I opened vial desktop and everything changed again. Plugging it back into a normal USB port and loading the saved layout restored expected behavior of the keys. This seems like a bug in Vial desktop that is changing the keymap when it detects a Via keyboard, even if it is not a Via keyboard. It does seem that the keys that break are all Vial specific features(or implementations of features).

sehrgut, 2025-11-14:
> [@xyzz](https://github.com/xyzz) Sure, but that doesn't really explain why the behavior of the keyboard changes.

If the KVM is altering the serial number, whether intentionally or unintentionally, it's likely altering keycodes as well. Does your KVM have some kind of "ISO/ANSI" compatibility mode that it might be invoking? This seems like a bug (or "undocumented feature") in the KVM.