vial-kb/vial-gui #309: VIAL File menu commands do not open dialog boxes on MacOS 15.4 Beta and Release candidate

fermat1313, 2025-03-24:
After installing MacOS 15.4 beta, File menu commands do not open the expected dialog boxes. Clicking on each does nothing. This has persisted through 3 beta versions and  in the RC that was released today.

StevenCurran, 2025-04-01:
also seeing this, very annoying. any ideas on fix

fermat1313, 2025-04-01:
Nope. I have a windows VM I can use to import/export, but it's a hindrance having to go there. I'm planning to revert my personal computer to 15.3 until this is worked out.

StevenCurran, 2025-04-01:
Its on GA release now, its not even beta / rc.

StevenCurran, 2025-04-01:
from system logs when opening the file window
`2025-04-01 19:34:13.008219-0400 0x107d41   Default     0x350bbf             34047  0    Vial: (libxpc.dylib) [com.apple.xpc:connection] [0x7fe8ffd088b0] invalidated because the current process cancelled the connection by calling xpc_connection_cancel()
2025-04-01 19:34:13.008376-0400 0x107d41   Error       0x350bbf             34047  0    Vial: (AppKit) [com.apple.AppKit:OpenSavePanels] Unable to display open panel (unrecognized error: Error Domain=com.apple.ViewBridge Code=14 "(null)" UserInfo={com.apple.ViewBridge.error.hint=-[NSViewServiceMarshal bootstrap:withReply:] caught NSInvalidArgumentException '*** -[NSUserDefaults setObject:forKey:inDomain:]: domain name cannot be nil or empty' with backtrace (
    "0   CoreFoundation                      __exceptionPreprocess + 176",
    "1   libobjc.A.dylib                     objc_exception_throw + 88",
    "2   Foundation                          -[NSUserDefaults(NSUserDefaults) setObject:forKey:inDomain:] + 420",
    "3   AppKit                              +[NSOSPDefaults _readFileListModeForIdentifier:inSaveMode:inDirectory:lastListForMediaBrowserMode:] + 524",
    "4   AppKit                              -[NSRemoteSavePanel(NSSavePanelLayout) _makeFinderKitView] + 416",
    "5   AppKit                              -[NSRemoteSavePanel(NSSavePanelLayout) _setupFileBrowserView] + 68",
    "6   AppKit                              -[NSRemoteSavePanel(NSSavePanelLayout) _initContentView] + 1544",
    "7   AppKit                              -[NSRemoteSavePanel initWithContentRect:styleMask:backing:defer:] + 488",
    "8   AppKit                              -[NSRemoteSavePanel initWithOptions:] + 476",
    "9   com.apple.appkit.xpc.openAndSavePan main + 708",
    "10  com.apple.appkit.xpc.openAndSavePan com.apple.appkit.xpc.openAndSavePan + 6440",
    "11  AppKit                              -[NSViewController _loadViewIfRequired] + 72",
    "12  AppKit                              __24-[NSViewController view]_block_invoke + 28",
    "13  AppKit                              NSPerformVisuallyAtomicChange + 108",
    "14  AppKit                              -[NSViewController view] + 160",
    "15  ViewBridge                          -[NSViewServiceMarshal _bootstrap:replyData:completion:] + 1308",
    "16  ViewBridge                          -[NSViewServiceMarshal bootstrap:withReply:] + 272",
    "17  CoreFoundation                      __invoking___ + 148",
    "18  CoreFoundation                      -[NSInvocation invoke] + 424",
    "19  CoreFoundation                      -[NSInvocation invokeWithTarget:] + 64",
    "20  ViewBridge                          -[NSVB_ViewServiceImplicitAnimationDecodingProxy forwardInvocation:] + 192",
    "21  CoreFoundation                      ___forwarding___ + 956",
    "22  CoreFoundation                      _CF_forwarding_prep_0 + 96",
    "23  CoreFoundation                      __invoking___ + 148",
    "24  CoreFoundation                      -[NSInvocation invoke] + 424",
    "25  CoreFoundation                      -[NSInvocation invokeWithTarget:] + 64",
    "26  ViewBridge                          -[NSVB_QueueingProxy forwardInvocation:] + 308",
    "27  CoreFoundation                      ___forwarding___ + 956",
    "28  CoreFoundation                      _CF_forwarding_prep_0 + 96",
    "29  CoreFoundation                      __invoking___ + 148",
    "30  CoreFoundation                      -[NSInvocation invoke] + 424",
    "31  CoreFoundation                      -[NSInvocation invokeWithTarget:] + 64",
    "32  CoreFoundation                      ___forwarding___ + 956",
    "33  CoreFoundation                      _CF_forwarding_prep_0 + 96",
    "34  CoreFoundation                      __invoking___ + 148",
    "35  CoreFoundation                      -[NSInvocation invoke] + 424",
    "36  ViewBridge                          __deferNSXPCInvocationOntoMainThread_block_invoke + 132",
    "37  ViewBridge                          __wrapBlockWithVoucher_block_invoke + 56",
    "38  ViewBridge                          deferredBlockOpportunity_block_invoke_2 + 356",
    "39  CoreFoundation                      __CFRUNLOOP_IS_CALLING_OUT_TO_A_BLOCK__ + 28",
    "40  CoreFoundation                      __CFRunLoopDoBlocks + 352",
    "41  CoreFoundation                      __CFRunLoopRun + 2384",
    "42  CoreFoundation                      CFRunLoopRunSpecific + 572",
    "43  HIToolbox                           RunCurrentEventLoopInMode + 324",
    "44  HIToolbox                           ReceiveNextEventCommon + 676",
    "45  HIToolbox                           _BlockUntilNextEventMatchingListInModeWithFilter + 76",
    "46  AppKit                              _DPSNextEvent + 684",
    "47  AppKit                              -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 688",
    "48  ViewBridge                          __77-[NSViewServiceApplication vbNextEventMatchingMask:untilDate:inMode:dequeue:]_block_invoke + 148",
    "49  ViewBridge                          -[NSViewServiceApplication _withToxicEventMonitorPerform:] + 152",
    "50  ViewBridge                          -[NSViewServiceApplication vbNextEventMatchingMask:untilDate:inMode:dequeue:] + 168",
    "51  ViewBridge                          -[NSViewServiceApplication nextEventMatchingMask:untilDate:inMode:dequeue:] + 100",
[... cut ...]
    "54  libxpc.dylib                        _xpc_objc_main + 816",
    "55  libxpc.dylib                        _xpc_main + 40",
    "56  libxpc.dylib                        _xpc_copy_xpcservice_dictionary + 0",
    "57  ViewBridge                          xpc_connection_handler + 0",
    "58  ViewBridge                          NSViewServiceMain + 360",
    "59  com.apple.appkit.xpc.openAndSavePan main + 156",
    "60  dyld                                start + 6000"
), com.apple.ViewBridge.error.description=NSViewBridgeErrorServiceBootstrap})`

xyzz, 2025-04-02:
fixed in 0.7.3

closed 2025-04-02 by xyzz

fermat1313, 2025-04-02:
Thank you for the quick fix!

StevenCurran, 2025-04-02:
Amazing thank you!