vial-kb/vial-gui #313: stucking on Preparing metadata?

kbladewht, 2025-04-04:
Any idea for help sort below out , it is stucking on Preparing metadata? 
I used python 3.12 by below fork, since I don't have python 3.6
https://github.com/edevil/vial-gui.git

pip install PyQt5==5.15.11 PyQt5-Qt5==5.15.2 PyQt5-sip==12.11.0
Collecting PyQt5==5.15.11
  Using cached PyQt5-5.15.11.tar.gz (3.2 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... -

xyzz, 2025-04-04:
python 3.6 is the only supported version

closed 2025-04-04 by xyzz

kbladewht, 2025-04-04:
3.12 also worked per https://github.com/edevil/vial-gui.git

[image]